debevv/nanoMODBUS #3: Add support to FC 43 / 14 (0x2B / 0x0E) Read Device Identification

debevv, 2022-01-24:


MoritzSchwoerer, 2023-08-08:
I am looking for this feature. Did someone already implemented read device identification?

debevv, 2023-08-08:
Nope, but it should be the next one I'm going to implement. Unfortunately no ETA though

Toroid82, 2024-02-07:
Hello. I recently added this functionality to my project. You can use the code below to add it to yours. Note that this is tested only with TCP connection, probably some tuning might be needed for serial. Only the mandatory partsa are implementented here, in other words only id code 1 is supported. See MODBUS Application Protocol Specification V1.1b chapter 6.21 for details. 

## nanomodbus.c
Add declaration:
```
static nmbs_error handle_read_device_id(nmbs_t* nmbs) {
    return handle_device_id(nmbs, nmbs->callbacks.device_id);
}
```
And function:
```
static nmbs_error handle_device_id(nmbs_t* nmbs, nmbs_error (*callback)(char*,char*,char*, void*)) {
	uint16_t length;

    nmbs_error err = recv(nmbs, 3);
    if (err != NMBS_ERROR_NONE)
        return err;

    uint16_t mei = nmbs->msg.buf[8];
    uint16_t id_code = nmbs->msg.buf[9];
    uint16_t obj_id = nmbs->msg.buf[10];

    NMBS_DEBUG_PRINT("id");

    err = recv_msg_footer(nmbs);
    if (err != NMBS_ERROR_NONE)
    	return err;

    if (mei != 0x0E)
    	return send_exception_msg(nmbs, NMBS_EXCEPTION_ILLEGAL_DATA_VALUE);

    if (id_code != 0x01)
    	return send_exception_msg(nmbs, NMBS_EXCEPTION_ILLEGAL_DATA_VALUE);

    if (obj_id != 0x00)
		return send_exception_msg(nmbs, NMBS_EXCEPTION_ILLEGAL_DATA_VALUE);

    if (callback) {
    	char vendorname[32] = {0};
    	char productcode[32] = {0};
    	char revision[32] = {0};
    	err = callback(vendorname, productcode, revision, nmbs->callbacks.arg);
    	if (err != NMBS_ERROR_NONE) {
    		if (nmbs_error_is_exception(err))
    			return send_exception_msg(nmbs, err);

    		return send_exception_msg(nmbs, NMBS_EXCEPTION_SERVER_DEVICE_FAILURE);
    	}

    	uint16_t vendorLen = strlen(vendorname);
    	uint16_t productLen = strlen(productcode);
    	uint16_t revisionLen = strlen(revision);

    	put_res_header(nmbs, vendorLen + productLen + revisionLen + 12);

    	put_1(nmbs, 0x0E);
    	put_1(nmbs, 1);
    	put_1(nmbs, 1);
    	put_1(nmbs, 0);
    	put_1(nmbs, 0);
    	put_1(nmbs, 3);
    	put_1(nmbs, 0);
    	put_1(nmbs, vendorLen);

    	memcpy(nmbs->msg.buf+nmbs->msg.buf_idx, vendorname, vendorLen);
    	nmbs->msg.buf_idx += vendorLen;

    	put_1(nmbs, 1);
    	put_1(nmbs, productLen);

    	memcpy(nmbs->msg.buf+nmbs->msg.buf_idx, productcode, productLen);
    	nmbs->msg.buf_idx += productLen;

    	put_1(nmbs, 2);
    	put_1(nmbs, revisionLen);

    	memcpy(nmbs->msg.buf+nmbs->msg.buf_idx, revision, revisionLen);
    	nmbs->msg.buf_idx += revisionLen;

    	err = send_msg(nmbs);
    	if (err != NMBS_ERROR_NONE)
    		return err;
    }
    else {
    	return send_exception_msg(nmbs, NMBS_EXCEPTION_ILLEGAL_FUNCTION);
    }

    return NMBS_ERROR_NONE;
}
```

## nanomodbus.h
Add a new callback definition inside nmbs_callbacks struct:
`nmbs_error (*device_id)(char* vendorname, char* productcode, char* revision, void* arg);`
This can be added in the end before void* arg;

## modbus_server.c
In your own server code register the callbacks, for example:
```
nmbs_callbacks callbacks = {0};
callbacks.read_holding_registers = handler_read_holding_registers;
callbacks.write_multiple_registers = handle_write_multiple_registers;
callbacks.device_id = handle_device_id;
```
And add your own server code what to write when a client asks for device id.
```
nmbs_error handle_device_id(char* vendorname, char* productcode, char* revision, void* arg) {
	strcpy(vendorname, "Mycompany");
	strcpy(productcode, "Mydevice");
	strcpy(revision, "0.0.1");

	return NMBS_ERROR_NONE;
}
```

debevv, 2024-02-10:
Implemented in a25ee5781aa5568108b538959a1c935d23c1deb0

closed 2024-02-10 by debevv