debevv/nanoMODBUS #5: Separate NMBS_ERROR_TRANSPORT into CRC error and others?

jonathangjertsen, 2022-05-10:
Hi, thanks for a very nice modbus implementation. One issue I've noticed is that the `NMBS_ERROR_TRANSPORT` includes several unrelated errors. In particular, it includes CRC mismatches. It would be nice to be able to separate this case from the others since this might indicate a problem with the modbus server rather than the client.

debevv, 2022-05-11:
Yes, I think I'd would be useful to add a `NMBS_ERROR_CRC`  and also something like `NMBS_ERROR_INVALID_TCP_MBAP` for this part at nanomodbus.c:317
```
        if (protocol_id != 0)
            return NMBS_ERROR_TRANSPORT;

        if (length > 255)
            return NMBS_ERROR_TRANSPORT;
```
This should cover all the transport-related error cases

debevv, 2022-05-12:
Added in 271f689771ca07f0a1863c9f54518b9ea9446936

closed 2022-05-12 by debevv