debevv/nanoMODBUS #29: Timing

AliMoal, 2023-06-10:
how this library handles 1.5 or 3.5 character times according to [Modbus over Serial Line](https://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf) document?

debevv, 2023-06-13:
Hi, basically it doesn't. It behaves similarly to libmodbus (https://github.com/stephane/libmodbus/commit/a11805cbfda23da36aed84bca3c5edc1413d664c and https://github.com/stephane/libmodbus/issues/18). We addressed this some time ago with another user in this [issue](https://github.com/debevv/nanoMODBUS/issues/26).  
I thought about implementing it at the time, but it would make the library more complex both internally and for the user. Also in my experience I never found a device honoring those timings, so having the library not rely on them makes it definitely more robust.  
Would you mind explaining me your use case?

AliMoal, 2023-06-14:
Actually, I want to use this library to communicate with PLC devices, I don't have any idea this lack of timing may cause problem or not.

AliMoal, 2023-06-14:
As I see in [here](https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c), libmodbus library handles those timings.

debevv, 2023-06-16:
Sorry, can you point me to the exact line? Because I can't find it in the linked file.  
By the way, if those PLC devices do follow the spec about the timings, there are two cases:
- when you receive data from them, there shouldn't be any problems because nanomodbus ignores the the intervals between characters and frames
-  when you send data, I think you can get away with waiting before making a request (if you are a client), waiting before responding in a request handler (if you are a server) and spacing the characters in your write() platform function

AliMoal, 2023-06-17:
I guess line 1277.
Could you show me how to implement Write() function? I'll use this library and see there is any problem or not.

debevv, 2023-06-21:
Yeah after reading my previous message again I realized that the `t1,5` in the spec is not mandatory inter-character spacing but instead a timeout, so you really don't need any particular handling in your write() function.  
The only things you should implement are the waits before requests and responses, to simulate inter-frame spacing

debevv, 2024-09-20:
Closing this, if you have more questions feel free to reopen the issue

closed 2024-09-20 by debevv