debevv/nanoMODBUS #30: Access to "Unit Identifier" byte in NMBS_TRANSPORT_TCP mode is impossible

arhiv6, 2023-06-26:
I wanted to use the library for creation modbus TCP to RTU gateway. But in the library impossible get access to "Unit identifier" byte. In function recv_msg_header() I see reciveing this byte: recive nmbs->msg.unit_id ([this string](https://github.com/debevv/nanoMODBUS/blob/44f319173566f01e56355c0031a75f6609b95b57/nanomodbus.c#L272)), but I can't access this field from outside the library.

arhiv6, 2023-06-26:
I find only one variant: I can use sometsing like as 

` nmbs_set_platform_arg(&nmbs, (void *) &nmbs);`

before calling nmbs_server_poll(&nmbs) and 

`nmbs_t *nmbs = (nmbs_t *) arg;`
`uint8_t unit_id = nmbs->msg.unit_id;`

in my server request callbacks. But [the documentation](https://debevv.github.io/nanoMODBUS/structnmbs__callbacks.html) says:

> All struct members are to be considered private, it is not advisable to read/write them directly.

It is only way?

debevv, 2023-06-26:
This was a bit of an oversight, it doesn't make sense to not expose the RTU unit ID in server callbacks. Can you try this branch https://github.com/debevv/nanoMODBUS/tree/unit_id_callback and check if it's enough?

arhiv6, 2023-06-26:
Yes, that's enough.

debevv, 2023-06-29:
Fixed in cbd4fa69fbc8be6d7a57a224f75f830ce16dd159

closed 2023-06-29 by debevv