debevv/nanoMODBUS #32: Not compiling with SDCC

DracoTomes, 2023-07-03:
Hi,

I'm using platformio to develop for an STM8 but nanoMODBUS is not compiling.

```Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 (2.1.0) > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz, 1KB RAM, 8KB Flash
DEBUG: Current (stlink) External (stlink)
PACKAGES:
 - framework-arduinoststm8 @ 0.50.210317
 - tool-stm8binutils @ 0.230.0 (2.30)
 - toolchain-sdcc @ 1.30901.11242 (3.9.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- nanoModbus
Building in release mode
Compiling .pio\build\stm8sblue\src\main.c.rel
Compiling .pio\build\stm8sblue\libb6b\nanoModbus\nanomodbus.c.rel
src\main.c:32: warning 85: in function read_serial unreferenced function argument : 'arg'
src\main.c:51: warning 85: in function write_serial unreferenced function argument : 'arg'
src\main.c:63: warning 85: in function handler_read_holding_registers unreferenced function argument : 'unit_id'
src\main.c:63: warning 85: in function handler_read_holding_registers unreferenced function argument : 'arg'
lib\nanoModbus\nanomodbus.c:1152: error 2: Initializer element is not a constant expression
lib\nanoModbus\nanomodbus.c:1290: error 0: Duplicate symbol 'subreq_file_number', symbol IGNORED
lib\nanoModbus\nanomodbus.c:1261: error 177: previously defined here
lib\nanoModbus\nanomodbus.c:1291: error 0: Duplicate symbol 'subreq_record_number', symbol IGNORED
lib\nanoModbus\nanomodbus.c:1262: error 177: previously defined here
lib\nanoModbus\nanomodbus.c:1292: error 0: Duplicate symbol 'subreq_record_length', symbol IGNORED
lib\nanoModbus\nanomodbus.c:1263: error 177: previously defined here
lib\nanoModbus\nanomodbus.c:1293: error 20: Undefined identifier 'subreq_record_length'
lib\nanoModbus\nanomodbus.c:1296: error 20: Undefined identifier 'subreq_file_number'
lib\nanoModbus\nanomodbus.c:1296: error 20: Undefined identifier 'subreq_record_number'
lib\nanoModbus\nanomodbus.c:1297: error 20: Undefined identifier 'subreq_record_length'
lib\nanoModbus\nanomodbus.c:1305: error 20: Undefined identifier 'subreq_record_length'
lib\nanoModbus\nanomodbus.c:1311: error 20: Undefined identifier 'subreq_record_length'
*** [.pio\build\stm8sblue\libb6b\nanoModbus\nanomodbus.c.rel] Error 1
========================================================================================= [FAILED] Took 10.02 seconds =========================================================================================
```

I'm new to C development and microcontrollers in general so I'm sorry if this is something obvious.

Any help is appreciated.

debevv, 2023-07-03:
Can you try building with the `--std-c99` compiler flag? I expect it to fix at least the errors about undefined identifiers

debevv, 2023-07-03:
I never used this compiler, so I made a couple of tests:
- with latest SDCC 4.3.0, almost all errors are gone, except for the one about VLAs (`Initializer element is not a constant expression`), which looks like SDCC is not supporting at all
- with your version, 3.9.0, I couldn't find any flag that could mitigate the problem, as per the documentation C99 support is incomplete  
  
So I would say, since all the errors are inside `handle_read_file_record()`, as a workaround you could disable it by defining `NMBS_SERVER_READ_FILE_RECORD_DISABLED`, if you don't plan on using FC 20  .
Otherwise this function should be substantially changed to remove the usage of features not supported by SDCC

DracoTomes, 2023-07-03:
The new new one builds with the c99 flag. The defines didn't do anything.

debevv, 2023-07-03:
> The new new one builds with the c99 flag.

Nice! Also I think the `--std-c99` was never necessary, so you can omit it

> The defines didn't do anything.

Strange, how are you defining them? I added `-DNMBS_SERVER_READ_FILE_RECORD_DISABLED -DNMBS_SERVER_WRITE_FILE_RECORD_DISABLED` to the sdcc command line and these functions were skipped as expected

DracoTomes, 2023-07-03:
I just put 
```c
#define NMBS_SERVER_READ_FILE_RECORD_DISABLED
#define NMBS_SERVER_WRITE_FILE_RECORD_DISABLED
```

into my main.c.

debevv, 2023-07-06:
ae6a93c963b497fc71296aa727ba09b40c15a2e7

closed 2023-07-06 by debevv

arhiv6, 2023-07-06:
@debevv,  I want to discuss about commit a08d59c. I think, it is non-optimal memory usage in `struct subreq`. Now, it usage 7*23=245 bytes on stack always. It may be big for embedded devices. It's a shame that almost always only the first element will be used (when we didn't use multiple sub-requests in one request). Maybe we can leave variable-length array (VLA)? VLA is part of C99 standard. Old versions SDCC were not supported VLA, but it fixed in [last year](https://sourceforge.net/p/sdcc/feature-requests/786/) and it already available in SDCC-4.3.0-RC1. Also, I can offer next variant:
```
struct {
        uint8_t reference_type;
        uint16_t file_number;
        uint16_t record_number;
        uint16_t record_length;
    }
#ifdef __STDC_NO_VLA__
    subreq[35];    // 245 / subreq_header_size
#else
    subreq[subreq_count];
#endif
```
Macros `__STDC_NO_VLA__` is is part of the standard С11, but old versions SDCC understand it too, including --std-c99.

What do you think?

debevv, 2023-07-09:
@arhiv6 Yes, that would be a downside of this approach. But consider that this is not the only place where it happens: the first one it comes to my mind is in `handle_read_discrete()` where a whole `nmbs_bitfield` is loaded onto the stack (or, in general, the `nmbs_bitfield` array itself when used by the user). For me it's a decent compromise between memory usage and ease of development.  
Anyway, in this case it doesn't make sense to not use your solution, I really didn't know the existence of these macros. I will implement it asap  
  
PS are you sure about SDCC 4.3.0 rc1? Because I tried 4.9.0 and the VLA didn't compile (see my previous comment)

arhiv6, 2023-07-09:
I checked VLA in SDCC 4.3.0-rc1 and it it not fully supported :( They added support VLA to their changelog and removed the line from the "SDCC Compiler User Guide" that VLA is not supported. I will send them a issue.  But this is not so important if you use a macro `__STDC_NO_VLA__`. Good news: I checked supporting macros `__STDC_NO_VLA__` for different SDCC versions. It supported since version 3.2 (released in July 2012) to last version 4.3 (released 07.07.2023).

Where did you get version 4.9.0? Maybe we are talking about different compilers? I checked it: [https://sdcc.sourceforge.net/](https://sdcc.sourceforge.net/).

debevv, 2023-07-11:
> @debevv, I want to discuss about commit [a08d59c](https://github.com/debevv/nanoMODBUS/commit/a08d59c5f0791be9b8fff0389cd9e62004f91c1f). I think, it is non-optimal memory usage in `struct subreq`. [...]

e3456986b551c891d4b95e89aaf5eded15e6f3a2