debevv/nanoMODBUS #36: Stack smashing protect failure

kopierreko, 2023-07-22:
Hi,

I'm still working with your library on a modbus project and in this case, I can see some perturbations on the bus (because a high frequency signal is running near from the modbus cable, but it can be filtered easily by the transceiver). I have a crash in my program (which is a little difficult to produce without all the code, but I can try in the future if really needed), localized at nanomodbus.c @1085 : if (!first_byte_received && err == NMBS_ERROR_TIMEOUT) :

`nmbs_error nmbs_server_poll(nmbs_t* nmbs) {
    msg_state_reset(nmbs);

    bool first_byte_received = false;
    nmbs_error err = recv_req_header(nmbs, &first_byte_received);
    if (err != NMBS_ERROR_NONE) {
        if (!first_byte_received && err == NMBS_ERROR_TIMEOUT)
            return NMBS_ERROR_NONE;

        return err;
    }`

The ESP32 returns "Stack smashing protect failure!" at the moment of the crash. 

`Decoding stack results
0x40083b81: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008bc45: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 121
0x40090eb5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x400ecf98: __stack_chk_fail at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/stack_check.c line 36
0x400e1daa: nmbs_server_poll at lib/nanomodbus/nanomodbus.c line 1085
0x400d40fe: ModBus::loop() at src/modbus.h line 358
0x400d50d7: loop() at src/main.cpp line 956
0x400e4685: loopTask(void*) at /Users/pierreko/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 50`

[image]

Do you have an idea ? 

Thanks 😉

debevv, 2023-07-27:
I can't see any place in `nmbs_server_poll()` where some stack smashing could take place. Usually that happens with operations like string manipulations, `memset()`s, arrays, etc. that could overwrite the stack but, from a cursory look, they aren't performed anywhere in the function or in its call tree.  
Can you check the function calling `nmbs_server_poll()`? Also, is `nmbs` a valid pointer?

kopierreko, 2023-07-27:
I founded a way to make it working by removing the glitches on the modbus wire, I had an electronic component which made EM perturbations but it suggests that the code can crash if there is too much perturbations on the bus. I can try what you’re talking about but not before 14/08.

Thanks for your answer and I will say you if I have this error 😊

debevv, 2023-07-27:
Can you try to reproduce the issue with `NMBS_DEBUG` enabled and show me the logs? Also, can you use the latest version (`v1.14.3`)?

closed 2024-02-10 by debevv

Silvesterrr, 2025-02-18:
@kopierreko could you provide the esp-idf example?

kopierreko, 2025-02-18:
> [@kopierreko](https://github.com/kopierreko) could you provide the esp-idf example?

Hi ! Sorry, I don't use est-idf in this project ^^