debevv/nanoMODBUS #57: Modbus TCP client over Wifi for Raspberry Pi Pico W (RP2040)

francdoc, 2024-07-16:
## Scope

The aim of this project is to implement the missing functionality needed to control the MODBUS TCP via Wifi interface on the Raspberry Pi [Pico W](https://www.raspberrypi.com/products/raspberry-pi-pico/)'s on board CYW43439 wifi/bluetooth chip. 

The way this is to be done is by implementing a port of read_fd_pico and write_fd_pico for this microcontroller. 

## Background

The development target is the RP2040 microcontroller using the C programming language that comes in the Pico W board package.

The work done so far has been achieved by implementing some lwIP functions using pico-examples as reference: [lwIP WIFI source exaples](https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/tcp_client). 

Any help would be greatly appreciated.

debevv, 2024-07-24:
Since you are on linux a good start would be checking out the linux code in the examples folder. Is there something in particular you'd like to know?

francdoc, 2024-07-25:
Hi @debevv, thank you for your answer. Yes, I checked them and I'm in the process of making a port of this [file](https://github.com/debevv/nanoMODBUS/blob/master/examples/linux/client-tcp.c) and this [file](https://github.com/debevv/nanoMODBUS/blob/master/examples/linux/platform.h) for rp2040. I was wondering if you could help me.

I explained in detail above the situation. The idea is to have a linux server running this [program ](https://github.com/debevv/nanoMODBUS/blob/master/examples/linux/server-tcp.c)and to have the rp2040 writing modbus tcp registers as a client so the server can read them.

I would appreciate any help you can give me making this port, I started the work [here](https://github.com/francdoc/nanoMODBUS/tree/rp2040-tcp-client/examples/pico/app).

francdoc, 2024-07-27:
Hi, I've just pushed a [Modbus RTU port](https://github.com/francdoc/nanoMODBUS/blob/rp2040-rtu-client/examples/pico/app/rtu-client.c) for the RP2040 (Pico). Check it out if you are interested.

debevv, 2024-09-20:
Merged in 7e4ef3e97e39664759e14bf993ff60d51dc8d850. Thank you

closed 2024-09-20 by debevv

francdoc, 2025-08-31:
Great, thank you