debevv/nanoMODBUS #58: How are non-contiguous registers managed if read with FC03?

effemmeffe, 2024-07-18:
I have a map of non-contiguous registers, I took a superficial look at the code, but I did not found if and how it's managed if the client requests a range that include some of my available registers, but also some of those that are unavailable.
Let say that I choose to provide holding registers 1-10 and 21-30 for a total of 20 holding registers.
What happens if the client sends a FC03 starting at register 5 for 10 consecutive registers?

pseudotronics, 2024-07-24:
It is left up to the individual port to implement address validation. nanoMODBUS handles the address validation in the sense that it checks to make sure the request is at least sane in regards to the specification (starting address, quantity). 

The standard says that FC03 is for reading contiguous blocks.

If you choose to only provide those registers in a non-contiguous block as described it would be on you to return `NMBS_EXCEPTION_ILLEGAL_DATA_ADDRESS` when this situation occurs, that is if you intend to meet the MODBUS standard.

pseudotronics, 2024-07-24:
The description for error code 02 actually provides a great example that mirrors what you asking exactly.

[image]

effemmeffe, 2024-07-24:
Yes, I was aware of the error code 02, that was exactly the point.
It just didn't occur to me that I could verify the range in the callback and since I saw the code already did some checks I was wondering.
Thanks.

closed 2024-09-20 by debevv