debevv/nanoMODBUS #60: Compilation issue

joegittingherdone, 2024-09-05:
Hello, 
I noticed in `nanomodbus.h` there is:

```
#ifndef NMBS_SLAVE_READ_COILS_DISABLED
    nmbs_error (*read_coils)(uint16_t address, uint16_t quantity, nmbs_bitfield coils_out, uint8_t unit_id, void* arg);
#endif
```

I assume it's supposed to be assigned in my code to `nmbs_read_coils()`, which is defined in `nanomodbus.c`.  The problem is that its parameters are incompatible:

```
nmbs_error nmbs_read_coils(nmbs_t* nmbs, uint16_t address, uint16_t quantity, nmbs_bitfield coils_out)
```
Am I supposed to add wrapper code or am I misunderstanding something here?

debevv, 2024-09-06:
Hi, the method you see under the `#define` is a function pointer member of the struct `nmbs_callbacks`. The function pointers inside of this struct are supposed to be assigned by the user with callback functions that are called when the **server** part of the library (slave) receives the corresponding request (in this case, FC 01). Refer to [this](https://github.com/debevv/nanoMODBUS/blob/67bd7b2075732c22fce5c89b5df3a21a4f5c8e12/examples/linux/server-tcp.c#L191) example for more details about server usage.  
The reason why all these methods are under `#define`s is to allow the user to disable them if not needed in order to reduce code size.

`nmbs_read_coils()` instead is a library function that the user can call to send a FC 01 request, when the library is used to implement a client device (master). You can see the usage of its sibling functions (like `nmbs_read_holding_registers()`) in the example in the README file.

joegittingherdone, 2024-09-06:
Hahaha, sorry, I forgot I changed all the client/server references to master/slave so my coworkers wouldn't be confused (we do PLC SCADA).  

I see what you're saying about the `nmbs_read_coils()` being for master/client mode.  Thank you for your answer, example, C library and all of your efforts!

closed 2024-09-06 by joegittingherdone