debevv/nanoMODBUS #71: Bug in all Server examples

Benygru, 2024-10-14:
This Bug is not in the stack, it is only in the server examples.

The Bug starts with this description. We have space for 32 Registers, but 0 to 32 are 33 Registers in number.
So only the index from 0 to 31 is supported for the array index.
Same for the coils.
This is only a comment but in the example code it is the same.

// The data model of this sever registers addresses from 0 to 32 
Bugfix: // The data model of this sever registers addresses from 0 to 31
#define REGS_ADDR_MAX 32

In this line we allow to access the Register index 32, so we are one index above the maximum Register which is 31.
if (address + quantity > REGS_ADDR_MAX + 1) {...} // Bug: This line of code allows to acces the index 32 -> out_of_range

I tested this bug with c++ code and used the .at(index) method. It has thrown std::out_of_range exception.

This code is the bugfix:
if (address + quantity > REGS_ADDR_MAX) {...} // Bug is fixed

Hope this helps
Best regards Benjamin

debevv, 2024-11-25:
Hi, yes the error was already reported by another user here https://github.com/debevv/nanoMODBUS/issues/39#issuecomment-1845427686, but i never fixed it. I will increase the array size by 1.

Thank you

debevv, 2024-11-25:
Fixed in b8d247f329a83cd4e12949a0a56e4c24228adca6

closed 2024-11-25 by debevv