debevv/nanoMODBUS #72: CMakeLists doesn't build a library

oedze, 2024-11-11:
The CMakeLists is now used to create a few executables with different settings.

It can't however be used as a static library right now which is usefull for embedded devices.

being able to do 

```txt
add_subdirectory(path/to/nanoMODBUS)

add_executable(myproject main.c)

target_link_libaries(myproject PRIVATE nanomodbus)
```

in our main projects CMakeLists.txt would be ideal for our scenario.

To be able to do this requires some change to the current CMakeLists.txt. If anyone's interested i can make a pull request.
If i do this i would also prefer to move the nanomodbus.h to a include folder.

donghoonpark, 2024-12-10:
How about supporting both approaches?
Adding ```nanomodbus``` as a library in CMake and compiling test examples with the library link would be the easiest way to achieve this.
Furthermore, for the project's CI, adopting a unit testing framework is also worth considering. If you don’t have time to implement it, I’d be happy to create a PR for it. Is there any particular reason you don’t want to make it a CMake-integrable library?

debevv, 2024-12-12:
No reason really, I just think that dropping the sources in the codebase is the simplest option. Anyway, if you find it useful, I will add a couple of lines in the CMakeLists to build a library.  
Regarding the tests, what do you mean with a unit testing framework?

debevv, 2024-12-12:
Added library target in a1705c0dff8e376c3ff45c73330bb3f1dbae2c18

donghoonpark, 2024-12-13:
I really appreciate your quick feedback. 
Library definition is quite useful when I use library with internet loading cmake command such as ```FetchContent```.
Regarding the unittest framework, to build test cases with minimal code and integrate them easily into the development environment, frameworks such as GoogleTest or Catch2 are commonly used, as they offer efficiency and compatibility.

donghoonpark, 2024-12-17:
```nanomodbus``` have no extra library. If you call ```add_subdirectory```, it gives you ```nanomodbus``` library spec. I already made a PR #79 that includes header file in library spec as well. Until the PR merge into master, you can refer the example [HERE](https://github.com/debevv/nanoMODBUS/blob/master/examples/stm32/CMakeLists.txt#L55).

patrick-dojofive, 2025-04-15:
#79 is merged so is this issue resolved?

debevv, 2025-04-16:
Yes, both static and dynamic libraries can be built, look at [CMakeLists.txt:11](https://github.com/debevv/nanoMODBUS/blob/a980f7f4c427e75203a740e84e30503edccd3e1e/CMakeLists.txt#L11C10-L11C27) for reference

closed 2025-04-16 by debevv