debevv/nanoMODBUS #73: Implementing timeouts with MCUXpresso SDK and LPUART FreeRTOS Driver

abusca, 2024-11-21:
Hi!
I'm using this library with MCUXpresso SDK's LPUART FreeRTOS Driver (API reference [here](https://mcuxpresso.nxp.com/api_doc/dev/3580/a00029.html)).
I implemented the read and write functions with LPUART_RTOS_Receive and LPUART_RTOS_Send functions. This driver also allows me to add timeouts with LPUART_RTOS_SetRxTimeout and LPUART_RTOS_SetTxTimeout functions (described [here](https://mcuxpresso.nxp.com/api_doc/dev/3580/a00029.html#ga3e211e0c1a2c93d9bdd73d6ca4639a10)).
These functions accept a constant timeout in ms (`rx_timeout_constant_ms`) and a per-byte timeout (`rx_timeout_multiplier_ms`). 

Then, LPUART FreeRTOS Driver's `LPUART_RTOS_Receive` and `LPUART_RTOS_Send` functions calculate the timeout like this:
`const TickType_t rxTickTimeout = (length * handle->rx_timeout_multiplier_ms + handle->rx_timeout_constant_ms) / portTICK_PERIOD_MS;`
...
Nonblocking send/receive and wait for events
...
`ev = xEventGroupWaitBits(handle->rxEvent, RTOS_LPUART_RX_COMPLETE | RTOS_LPUART_RING_BUFFER_OVERRUN | RTOS_LPUART_HARDWARE_BUFFER_OVERRUN, pdTRUE, pdFALSE, (rxTickTimeout > 0u) ? rxTickTimeout : portMAX_DELAY);`

So, as you can see, when timeout is less or equal to 0 it waits forever.

My questions are:
In order to adapt `byte_timeout_ms` to my timeout functions, should I assign `byte_timeout_ms` to `rx_timeout_multiplier_ms` and keep `rx_timeout_constant_ms` to 0?
When `byte_timeout_ms` is 0 it should expire, but how can I do this? Can I just add a timeout of 1ms to `rx_timeout_constant_ms` in this case?

Thank you!

debevv, 2024-11-25:
Hi,
- I guess? `byte_timeout_ms` was conceived with a primitive byte-by-byte read loop in mind, where you reset a timeout every time before receiving a byte. What you are doing here looks like the closest thing to a byte timeout so i guest that's the way
- 0 means that you try to read once and then immediately return whatever you read. So yes, if that API does not support this behavior, put the smallest timeout you can to simulate it

abusca, 2024-12-01:
Thank you @debevv !

closed 2024-12-01 by abusca