debevv/nanoMODBUS #84: Answer to frame with incorrect ID

Vyasem, 2025-03-06:
Hi, when a device  receives a frame with an incorrect ID and function code, it will still send an answer as an exception
If the ID is incorrect, then nanomodbus will label this frame as ignored and will use this information only  for subfunction, not for exception function.
I suggest returning an "NMBS_ERROR_INVALID_UNIT_ID" error before calling the "handle_req_fc" function if an invalid ID is provided.
Thanks.

debevv, 2025-03-06:
Hi, sorry but from a quick review of the code looks like the `ignored` flag is checked every time we are going to send a response. But maybe it's just me? Can you point me where the code does not follow this behavior?

Vyasem, 2025-03-07:
Hi, thanks for your reply.

Vyasem, 2025-03-07:
Hi, thanks for your reply.
The "send_exception_msg" function sends a message, even if the ID is incorrect.
I have multiple devices connected to the bus, and I use the Modbus bootloader. When one device is flashing its firmware, the others send an exception message on the bus.
After that, the flashing firmware crashed.

debevv, 2025-03-07:
I don't think this is happening because of incorrect ID/FC are received. `send_exception_msg` (and any other response) is always called only if `!nmbs->msg.ignored`

alexandration, 2025-03-07:
@debevv thanks for your reply. The reason for this behavior is that we are using the latest [release](https://github.com/debevv/nanoMODBUS/releases/tag/v1.19.0), not the latest commit on master. Indeed, you have already solved this problem.

closed 2025-03-07 by Vyasem

debevv, 2025-03-07:
> [@debevv](https://github.com/debevv) thanks for your reply. The reason for this behavior is that we are using the latest [release](https://github.com/debevv/nanoMODBUS/releases/tag/v1.19.0), not the latest commit on master. Indeed, you have already solved this problem.

Well, this is bad anyway. User are supposed to use the latest release, not master. I'm making a release right away. Thank you