debevv/nanoMODBUS #88: TCP handling consumes flash space even when not used

Avamander, 2025-05-04:
It seems like the TCP logic still consumes flash space even when not used (compiler isn't optimizing it out).

When I commented out sections that handled TCP cases in `recv_msg_header`, `put_msg_header`, `nmbs_create`, `set_msg_header_size`, `recv_res_header` and `nmbs_strerror`, it saved me more than a hundred bytes. It might be useful to wrap these between an ifdef to make it a bit easier. (Depends on how worth it it is for others I guess.)

That and a few other steps took the server example down below the 8k line for me, which at least lets me run it. (But it's still consuming a bit too much space to properly fit on an ATTiny817, so I'm still looking for ways to trim it further.)

debevv, 2025-05-06:
Hi, are you using the defines described in the readme? https://github.com/debevv/nanoMODBUS?tab=readme-ov-file#misc  
Are you enabling size optimizations in the compiler?

Avamander, 2025-05-07:
Yes, I've used the defines (and it does help) and the compiler is set to `-Os`. It just seems like those unreachable cases aren't optimized out (entirely) by avr-gcc.

debevv, 2025-05-09:
What defines are you using? So i can reproduce your situation

Avamander, 2025-05-11:
I'm using `NMBS_CLIENT_DISABLED`, `NMBS_STRERROR_DISABLED`, `NMBS_SERVER_READ_DEVICE_IDENTIFICATION_DISABLED`. I honestly need a very minimal server (and a client). Though after trying to squeeze everything together, I still think I'm lacking space on the ATTiny817 to actually end up using it for this project (as a simple MODBUS sensor).

In any case it kinda makes sense that in such cases functions like `recv_msg_header` is included in the resulting binary and also why the compiler might not be able to remove the TCP section.  (Because `nmbs->platform.transport == NMBS_TRANSPORT_TCP` is a data-dependant conditional?)

debevv, 2025-05-21:
I tried to manually remove TCP-related code, but as expected it doesn't help much. I think you can get away, if it's possible in your project, with using only a couple of function codes and excluding all the others, like this:
```
#define NMBS_CLIENT_DISABLED
#define NMBS_STRERROR_DISABLED

#define NMBS_SERVER_READ_COILS_DISABLED
#define NMBS_SERVER_READ_DISCRETE_INPUTS_DISABLED
#define NMBS_SERVER_READ_INPUT_REGISTERS_DISABLED
#define NMBS_SERVER_WRITE_SINGLE_COIL_DISABLED
#define NMBS_SERVER_WRITE_SINGLE_REGISTER_DISABLED
#define NMBS_SERVER_WRITE_MULTIPLE_COILS_DISABLED
#define NMBS_SERVER_READ_FILE_RECORD_DISABLED
#define NMBS_SERVER_WRITE_FILE_RECORD_DISABLED
#define NMBS_SERVER_READ_WRITE_REGISTERS_DISABLED
#define NMBS_SERVER_READ_DEVICE_IDENTIFICATION_DISABLED
```
with these defines only `read_holding_registers` and `write_multiple_registers` are supported, which should be enough for a simple sensor.
`readelf`, on a binary built with `avr-gcc` in release mode and `-Os`, reports a very small code size:
```
  [N°] Nome              Tipo            Indir    Off    Dimen  ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .text             PROGBITS        00000000 000074 001618 00  AX  0   0  2
  [ 2] .data             PROGBITS        00800060 00168c 000000 00  WA  0   0  1
  [ 3] .comment          PROGBITS        00000000 00168c 000011 01  MS  0   0  1
  [ 4] .shstrtab         STRTAB          00000000 001f46 000030 00      0   0  1
  [ 5] .symtab           SYMTAB          00000000 0016a0 000470 10      6  26  4
  [ 6] .strtab           STRTAB          00000000 001b10 000436 00      0   0  1

  Num:    Valore Dim  Tipo    Assoc  Vis      Ind Nome
     0: 00000000     0 NOTYPE  LOCAL  DEFAULT  UND 
     1: 00000000     0 SECTION LOCAL  DEFAULT    1 .text
     2: 00800060     0 SECTION LOCAL  DEFAULT    2 .data
     3: 00000000     0 SECTION LOCAL  DEFAULT    3 .comment
     4: 00000000     0 FILE    LOCAL  DEFAULT  ABS nanomodbus.c
     5: 0000003e     0 NOTYPE  LOCAL  DEFAULT  ABS __SP_H__
     6: 0000003d     0 NOTYPE  LOCAL  DEFAULT  ABS __SP_L__
     7: 0000003f     0 NOTYPE  LOCAL  DEFAULT  ABS __SREG__
     8: 00000000     0 NOTYPE  LOCAL  DEFAULT  ABS __tmp_reg__
     9: 00000001     0 NOTYPE  LOCAL  DEFAULT  ABS __zero_reg__
    10: 00000000    46 FUNC    LOCAL  DEFAULT    1 get_2
    11: 0000002e    36 FUNC    LOCAL  DEFAULT    1 put_2
    12: 00000052    44 FUNC    LOCAL  DEFAULT    1 flush
    13: 000000ec   188 FUNC    LOCAL  DEFAULT    1 put_msg_header
    14: 000001a8   228 FUNC    LOCAL  DEFAULT    1 send_msg
    15: 0000028c    76 FUNC    LOCAL  DEFAULT    1 send_exception_msg
    16: 000002d8   146 FUNC    LOCAL  DEFAULT    1 recv
    17: 0000036a   106 FUNC    LOCAL  DEFAULT    1 recv_msg_footer
    18: 000003d4   598 FUNC    LOCAL  DEFAULT    1 recv_msg_header.[...]
    19: 0000062a   310 FUNC    LOCAL  DEFAULT    1 recv_res_header
    20: 00000000     0 FILE    LOCAL  DEFAULT  ABS client_disabled.c
    21: 0000003e     0 NOTYPE  LOCAL  DEFAULT  ABS __SP_H__
    22: 0000003d     0 NOTYPE  LOCAL  DEFAULT  ABS __SP_L__
    23: 0000003f     0 NOTYPE  LOCAL  DEFAULT  ABS __SREG__
    24: 00000000     0 NOTYPE  LOCAL  DEFAULT  ABS __tmp_reg__
    25: 00000001     0 NOTYPE  LOCAL  DEFAULT  ABS __zero_reg__
    26: 00000ecc    34 FUNC    GLOBAL DEFAULT    1 nmbs_callbacks_create
    27: 00000760   148 FUNC    GLOBAL DEFAULT    1 nmbs_create
    28: 00000968   300 FUNC    GLOBAL DEFAULT    1 recv_read_file_r[...]
    29: 0000ffa0     0 NOTYPE  GLOBAL DEFAULT  ABS __DATA_REGION_LE[...]
    30: 00000c34   664 FUNC    GLOBAL DEFAULT    1 recv_read_device[...]
    31: 00000000     0 NOTYPE  GLOBAL DEFAULT  ABS __TEXT_REGION_OR[...]
    32: 00000000     0 NOTYPE  GLOBAL DEFAULT    1 __trampolines_start
    33: 00001618     0 NOTYPE  GLOBAL DEFAULT    1 _etext
    34: 0000084e    14 FUNC    GLOBAL DEFAULT    1 nmbs_set_platform_arg
    35: 00000eee   146 FUNC    GLOBAL DEFAULT    1 nmbs_server_create
    36: 00000a94   416 FUNC    GLOBAL DEFAULT    1 recv_write_file_[...]
    37: 000015e0    22 FUNC    GLOBAL DEFAULT    1 memcpy
    38: 00001618     0 NOTYPE  GLOBAL DEFAULT  ABS __data_load_end
    39: 00001516    10 FUNC    GLOBAL DEFAULT    1 write_empty
    40: 00000000     0 NOTYPE  GLOBAL DEFAULT    1 __trampolines_end
[... 6 lines cut ...]
    47: 00000806    18 FUNC    GLOBAL DEFAULT    1 nmbs_set_byte_timeout
    48: 000007f4    18 FUNC    GLOBAL DEFAULT    1 nmbs_set_read_timeout
    49: 0000007e   110 FUNC    GLOBAL DEFAULT    1 nmbs_crc_calc
    50: 00810000     0 NOTYPE  GLOBAL DEFAULT    2 __eeprom_end
    51: 000015f6    34 FUNC    GLOBAL DEFAULT    1 strncpy
    52: 000015c2    30 FUNC    GLOBAL DEFAULT    1 memcmp
    53: 00000400     0 NOTYPE  GLOBAL DEFAULT  ABS __SIGNATURE_REGI[...]
    54: 0000085c   132 FUNC    GLOBAL DEFAULT    1 recv_write_multi[...]
    55: 00000000     0 NOTYPE  GLOBAL DEFAULT    1 __ctors_start
    56: 0000152a   152 FUNC    GLOBAL DEFAULT    1 main
    57: 00800060     0 NOTYPE  GLOBAL DEFAULT  ABS __DATA_REGION_OR[...]
    58: 00000818    42 FUNC    GLOBAL DEFAULT    1 nmbs_platform_co[...]
    59: 00000400     0 NOTYPE  GLOBAL DEFAULT  ABS __USER_SIGNATURE[...]
    60: 000008e2     2 FUNC    GLOBAL DEFAULT    1 recv_write_singl[...]
    61: 00000000     0 NOTYPE  GLOBAL DEFAULT    1 __dtors_start
    62: 00000000     0 NOTYPE  GLOBAL DEFAULT    1 __ctors_end
[... cut ...] ta
    64: 00800060     0 NOTYPE  GLOBAL DEFAULT    2 _end
    65: 00000842    12 FUNC    GLOBAL DEFAULT    1 nmbs_set_destina[...]
    66: 000008e0     2 FUNC    GLOBAL DEFAULT    1 recv_write_singl[...]
    67: 00010000     0 NOTYPE  GLOBAL DEFAULT  ABS __EEPROM_REGION_[...]
    68: 000008e4   132 FUNC    GLOBAL DEFAULT    1 recv_write_multi[...]
    69: 00000400     0 NOTYPE  GLOBAL DEFAULT  ABS __FUSE_REGION_LE[...]
    70: 00002000     0 NOTYPE  GLOBAL DEFAULT  ABS __TEXT_REGION_LE[...]
```

Avamander, 2025-05-25:
> I tried to manually remove TCP-related code, but as expected it doesn't help much. I think you can get away, if it's possible in your project, with using only a couple of function codes and excluding all the others, like this:

It depends, the ~100-150 bytes reduction was almost exactly what I needed to at least try out the examples. Though I think upgrading the microcontroller to larger storage one would be significantly more economical at this point and I'm probably going to do that.

closed 2026-02-01 by debevv