debevv/nanoMODBUS #91: Hello, I followed your STM32 example to port nanomodbus to the STM32F103 microcontroller

zhangjingliu, 2025-05-12:
Hello, I followed your STM32 example to port nanomodbus to the STM32F103 microcontroller, and then tested it using the Modbus Poll master software, but it keeps showing a timeout error, and it’s never correct. Could you turn the STM32's nanomodbus into a complete example? I noticed there is a bug in this example! Can you help fix this bug?

zhangjingliu, 2025-05-12:
[Modbus Slave.zip](https://github.com/user-attachments/files/20158311/Modbus.Slave.zip)
this is my project, it didn't work , always show error time 

[image]

donghoonpark, 2025-05-22:
Your code looks like implemented modbus client (modbus master) and the modbus poll is master simulator as well. It looks like you check your configuration first.