debevv/nanoMODBUS #94: Question: Using this library with AVR and TCP

Kampi, 2025-06-19:
Hi,

I want to use this library with a Wiznet module for Modbus TCP (Client only). You already provide an example for STM32. Can you please tell me where this Wiznet header comes from?

Thanks!

donghoonpark, 2025-06-19:
I linked wiznet library as cmake module and intermediate porting layer is written in [my another repository](https://github.com/donghoonpark/wizchip-cmake). Porting would be easy if you know how to replace stm32 spi function to avr's one.

Kampi, 2025-06-19:
Thanks! That shouldn't be an issue :)

Kampi, 2025-06-19:
One last question: Can you recommend a specific chip / module?

closed 2025-06-19 by Kampi

Kampi, 2025-06-23:
Sure - do you have an example for a simple ping from my PC (or Raspberry Pi) to the chip?

donghoonpark, 2025-06-24:
I'd like to recommend you to implement loopback application first using the functions implemented in ```loopback.c```
Wiznet used [this program (AX2)](https://iotedition.tistory.com/7) for their network performance test. You can check the performance you can expect by using it.

Kampi, 2025-06-25:
Hi @donghoonpark,
Thanks again for the tip. AX2 looks a bit weird to me, and it throws some unknown errors. I´ve tested it with Hercules and a loopback code and a TCP server on my XMega and a Python application. Both are functional, so I can start with the Modbus integration